edufolly/flutter_bluetooth_serial #139: type 'Future<dynamic>' is not a subtype of type 'FutureOr<int>' in type cast

hmney, 2021-07-09:
### Problem summary
Hello folks, I will explain what's happened to me (you can check first my environment before continue reading). I tried to get my Bluetooth state from the state getter in the file FlutterBluetoothSerial.dart, and I got this exception `type 'Future<dynamic>' is not a subtype of type 'FutureOr<int>' in type cast`. So I went to the file and I tried this code 
https://github.com/edufolly/flutter_bluetooth_serial/blob/50bf77837d7956c13b0611c298a8041d524114fa/lib/FlutterBluetoothSerial.dart#L51-L52
 and I get my state.

So the problem was solved with me but I want to know if I'm wrong in anything here. by the way this my first essay for me to contribute in open source, forgive me if I'm wrong in anything.

this is my commit to fix the problem: https://github.com/edufolly/flutter_bluetooth_serial/commit/b2c40234655ca23c5865ac167aa2db9dc98d0341

if I'm right maybe the same problem will be happening here: https://github.com/edufolly/flutter_bluetooth_serial/blob/50bf77837d7956c13b0611c298a8041d524114fa/lib/FlutterBluetoothSerial.dart#L94-L98

### Code to test
https://gist.github.com/hmney/b4d0bb3e74c9675b247bb82a60eff95f

### Environment

`Flutter 2.2.2 • channel stable • https://github.com/flutter/flutter.git
Framework • revision d79295af24 (4 weeks ago) • 2021-06-11 08:56:01 -0700
Engine • revision 91c9fc8fe0
Tools • Dart 2.13.3`

Afonsocraposo, 2021-07-10:
I also get this error with `getBondStateForAddress`.

nicolasjon, 2021-07-10:
The current example code does not work because of this change in 0.3.0
from

Future<BluetoothState> get state async => BluetoothState.fromUnderlyingValue( 
       await (_methodChannel.invokeMethod('getState')));

To

Future<BluetoothState> get state async => BluetoothState.fromUnderlyingValue( 
       await (_methodChannel.invokeMethod('getState') as FutureOr<int>));