edufolly/flutter_bluetooth_serial #146: Connect error, cant connect to device

awaisjamil-vd, 2021-09-08:
java.io.IOException: read failed, socket might closed or timeout, read ret: -1
	at android.bluetooth.BluetoothSocket.readAll(BluetoothSocket.java:970)
	at android.bluetooth.BluetoothSocket.readInt(BluetoothSocket.java:984)
	at android.bluetooth.BluetoothSocket.connect(BluetoothSocket.java:536)
	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:57)
	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:64)
	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler.lambda$onMethodCall$4$FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler(FlutterBluetoothSerialPlugin.java:1007)
	at io.github.edufolly.flutterbluetoothserial.-$$Lambda$FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler$Q4kMxByjf_xgX66iy8HZNeJ0z8I.run(Unknown Source:10)
	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:305)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
	at java.lang.Thread.run(Thread.java:923)

vaibhiarora03, 2021-10-04:
@awaisjamil-vd did you find any solution for this?

awaisjamil-vd, 2021-10-04:
> @awaisjamil-vd did you find any solution for this?

Nope, I had to write whole bluetooth module in native android and then used it with flutter.

vaibhiarora03, 2021-10-04:
Oh, okay. Can you share some sample or something which could help ease the integration for me and others as well.

awaisjamil-vd, 2021-10-04:
> Oh, okay. Can you share some sample or something which could help ease the integration for me and others as well.

class BluetoothChatUtils {
    private val bluetoothAdapter: BluetoothAdapter
    private var connectThread: ConnectThread? = null
    private var acceptThread: AcceptThread? = null
    private var connectedThread: ConnectedThread? = null
    private val APP_UUID = UUID.fromString("fa87c0d0-afac-11de-8a39-0800200c9a66")
    private val APP_NAME = "BluetoothChatApp"
    private val executorService = Executors.newSingleThreadExecutor()

    @set:Synchronized
    var state: Int
    private val STATE_NONE = 0
    private val STATE_LISTEN = 1
    private val STATE_CONNECTING = 2
    private val STATE_CONNECTED = 3

    private val FLAG_MSG = 0

    companion object {
        private var handler: Handler? = null

        @Volatile
        private var instance: BluetoothChatUtils? = null

        fun getInstance(handler: Handler?): BluetoothChatUtils? {
            this.handler = handler
            if (instance == null) {
                synchronized(BluetoothChatUtils::class.java) {
                    if (instance == null) {
                        instance = BluetoothChatUtils()
                    }
                }
            }
            return instance
        }
    }

    init {
        state = STATE_NONE
        bluetoothAdapter = BluetoothAdapter.getDefaultAdapter()
    }

    @Synchronized
    private fun start() {
        if (connectThread != null) {
            connectThread!!.cancel()
            connectThread = null
        }
        if (acceptThread == null) {
            acceptThread = AcceptThread()
            acceptThread!!.start()
        }
        if (connectedThread != null) {
            connectedThread!!.cancel()
            connectedThread = null
        }
        state = STATE_LISTEN
    }

    @Synchronized
    fun stop() {
        if (connectThread != null) {
            connectThread!!.cancel()
            connectThread = null
        }
        if (acceptThread != null) {
            acceptThread!!.cancel()
            acceptThread = null
[... 176 lines cut ...]
        this@BluetoothChatUtils.start()
    }

    @Synchronized
    private fun connectionFailed() {
        handler!!.obtainMessage(MainActivity.BLUE_TOOTH_TOAST, -1, -1, "Cant connect to the device").sendToTarget()
        this@BluetoothChatUtils.start()
    }

    @Synchronized
    private fun connected(socket: BluetoothSocket?, device: BluetoothDevice) {
        if (connectThread != null) {
            connectThread!!.cancel()
            connectThread = null
        }
        if (connectedThread != null) {
            connectedThread!!.cancel()
            connectedThread = null
        }
        connectedThread = ConnectedThread(socket!!)
        connectedThread!!.start()
        handler!!.obtainMessage(MainActivity.BLUE_TOOTH_SUCCESS, -1, -1, device).sendToTarget()
        state = STATE_CONNECTED
    }
}

vaibhiarora03, 2021-10-04:
@awaisjamil-vd  thanks.

amiruljack, 2021-11-15:
have you update this?

vaibhiarora03, 2021-11-15:
no. I haven't.

BrenoNaSa, 2021-11-18:
Hi @awaisjamil-vd, 

I would like to know an example of futter application.  
How are you using this native class?
If there is any flutter project at github, could you share with me?

i'm looking forward to hearing from you.

sasikdar, 2021-11-24:
Hi @awaisjamil-vd ,

is it possible for you to share an example of how you are using the andriod library with flutter?

vaibhiarora03, 2022-02-09:
Hi @awaisjamil-vd ,

Can you please tell me what was the occurance rate of this issue using this library? 

After integrating your library was this issue completely resolved for you?

Thanks

vaibhiarora03, 2022-02-09:
@sasikdar, @amiruljack , @BrenoNaSa is this issue resolved for you?

amiruljack, 2022-02-22:
> @sasikdar, @amiruljack , @BrenoNaSa is this issue resolved for you?

no.

sasikdar, 2022-02-22:
I did not check if there are new fixes. But it did not work for me. I ended up using Flutter_nearby. There is a similar package from Huwaeii too :-)

vaibhiarora03, 2022-02-22:
@amiruljack okay.

vaibhiarora03, 2022-02-22:
@sasikdar okay.

udit6023, 2023-06-11:
@sasikdar were you able to solve this connection issue using this flutter_nearby package?