edufolly/flutter_bluetooth_serial #154: read failed, socket might closed or timeout, read ret: -1

HajerWael, 2021-11-22:
Always getting this error when try to connect to a paired device (android mobile).

vaibhiarora03, 2022-02-04:
Hi @HajerWael. Were you able to solve this error?

martesabt, 2022-02-09:
Same here, in my case i'm tying to connect to an Intel NUC.

vaibhiarora03, 2022-02-09:
Hi @martesabt. What's the occurance rate of this error for you? Is it random or you are stuck with it?

vaibhiarora03, 2022-02-09:
Hi @martesabt . 

I did some searching for this issue and had implemented it from those learnings. You can have a look at my implementation. Though the code I tried didn't work for me but worked for many others. Maybe it can help you out. 
https://stackoverflow.com/questions/71038323/ioexception-read-failed-socket-might-closed-or-timeout-read-ret-1

My source:- https://stackoverflow.com/questions/18657427/ioexception-read-failed-socket-might-closed-bluetooth-on-android-4-3

martesabt, 2022-02-09:
Thanks @vaibhiarora03 I will take a look at it.

I noticed you closed #160 did you found an alternative?

This is frustrating :(

vaibhiarora03, 2022-02-09:
@martesabt The logs which i mentioned in the post had my custom logs in it making it useless to point to the exact issue. Thats why closed it and opened a new issue with proper information.

#164

martesabt, 2022-02-09:
@vaibhiarora03 Oh, ok!

I hope we can solve this!

RRohitM, 2022-06-17:
hello Everyone 
I have **read failed, socket might be closed or timeout read ret: -1**  this issues you all are using static UUID for the connection so that you are facing this issue, You can use your Bluetooth device UUID for the connection and you can resolve this issue.

You can change BluetoothConnection.java class 
you can replace connect method

 /// Connects to given device by hardware address (default UUID used)
    public void connect(String address) throws IOException {
        BluetoothAdapter btAdapter = BluetoothAdapter.getDefaultAdapter();
        BluetoothDevice device = btAdapter.getRemoteDevice(address);
        ParcelUuid[] uuids = (ParcelUuid[]) device.getUuids();
        connect(address, uuids[0].getUuid());
    }

guigzp, 2022-09-27:
> hello Everyone I have **read failed, socket might be closed or timeout read ret: -1** this issues you all are using static UUID for the connection so that you are facing this issue, You can use your [...]

this saved me, thx man

MrAnderson2001, 2023-01-29:
@RRohitM Where do you change that part of the code? I am using the example codes from the flutter_bluetooth_serial package. I can find the devices, but I cannot connect to them.

MrAnderson2001, 2023-01-29:
@RRohitM I tried this method, but it caused my code to compile to my device. Were there any other changes that you made to the base example codes?

mohsinnaqvi0606, 2023-09-18:
I tried @RRohitM  solution but still same issue

ugifractal, 2024-01-14:
This should help. https://stackoverflow.com/a/25647197/5552022

leverkusen188, 2024-09-24:
> This should help. https://stackoverflow.com/a/25647197/5552022
> 
> So change from
> 
> ```java
> device.createRfcommSocketToServiceRecord(uuid); // or device.createInsecureRfcommSocketToServiceRecord(uuid);
> ```
> 
> to
> 
> ```java
> socket = (BluetoothSocket) device.getClass().getMethod("createRfcommSocket", new Class[] {int.class}).invoke(device, 1);
> ```

Saved my life man!

ceperman, 2024-11-12:
It's worth pointing out that the corresponding server socket needs to be active when `connect` is issued, or this exception will result. The docs say that `connect` is a blocking call but whatever it may be blocking for, it's not waiting for the server to be active.

This is a problem for peer-to-peer communication. If you don't know when or if the server will be active, as a client you just have to handle this exception and repeat trying. I wasted a lot time discovering this.