edufolly/flutter_bluetooth_serial #158: Run Flutter bluetooth serial in background service. Bluetooth unavailable

PhamHoangBao, 2022-01-09:
Hi everyone,
I am currently developing a android app that can scan bluetooth several time even when I exit my app (not closed). So I combined flutter_bluetooth_serial with flutter_workmanager to send it on background job. I changed _startDicorvery() function in file DiscoveryPage.dart as the follow image
[image]
and write a callback function in WorkManager class as the follow image:
[image]
However, when I turn on bluetooth and execute this function, background job cannot detect bluetooth and throw error:
[image]
Does anyone can recommend me how to resolve this issue or anyone has worked with bluetooth scanning in background job can share me your experiences ?
Thank you so much.

Baluds, 2022-05-07:
I am also having same issue, did you get any idea?

rdmasaile, 2023-04-30:
Is there any solution to this?