edufolly/flutter_bluetooth_serial #161: Android 12 support

calmh, 2022-02-03:
### Problem summary

It appears the library doesn't currently support the permissions model in Android 12.

### Steps to reproduce

On an Android 12 phone, Bluetooth permissions are not requested and Bluetooth devices cannot be enumerated.

### Environment

(TBD we're still investigating. Will try to patch once we've narrowed down where and how.)

calmh, 2022-03-03:
I forgot to mention it here, but there's a fix in #163 that we've had in production now for a while.

pst9354, 2022-04-01:
> I forgot to mention it here, but there's a fix in #163 that we've had in production now for a while.

Any word from the author, to include this fix and make a new release?

rrrovalle, 2022-04-04:
News about #163 release? I really need to use it on Android 12

rrrovalle, 2022-04-04:
I'll do the same! Thank's for the reply man :-)

pst9354, 2022-04-04:
You can use our repo in the meantime, just add this to your pubspec.yaml:

#  flutter_bluetooth_serial: ^0.4.0

  flutter_bluetooth_serial:
    git:
      url: https://github.com/pst9354/flutter_bluetooth_serial
      ref: master

Catalunya69, 2022-06-17:
I have tried it, and I have this problem, can you help me? Thank 

`D:\Flutter\src\.pub-cache\git\flutter_bluetooth_serial-49d778bd52b9c29cb1105190ff05c6dd6f2b8617\android\src\main\java\io\github\edufolly\flutterbluetoothserial\FlutterBluetoothSerialPlugin.java:459: error: cannot find symbol
        if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.S) {
                                                        ^
  symbol:   variable S
  location: class VERSION_CODES
D:\Flutter\src\.pub-cache\git\flutter_bluetooth_serial-49d778bd52b9c29cb1105190ff05c6dd6f2b8617\android\src\main\java\io\github\edufolly\flutterbluetoothserial\FlutterBluetoothSerialPlugin.java:462: error: cannot find symbol
                            Manifest.permission.BLUETOOTH_SCAN)
                                               ^
  symbol:   variable BLUETOOTH_SCAN
  location: class permission
D:\Flutter\src\.pub-cache\git\flutter_bluetooth_serial-49d778bd52b9c29cb1105190ff05c6dd6f2b8617\android\src\main\java\io\github\edufolly\flutterbluetoothserial\FlutterBluetoothSerialPlugin.java:465: error: cannot find symbol
                            Manifest.permission.BLUETOOTH_CONNECT)
                                               ^
  symbol:   variable BLUETOOTH_CONNECT
  location: class permission
D:\Flutter\src\.pub-cache\git\flutter_bluetooth_serial-49d778bd52b9c29cb1105190ff05c6dd6f2b8617\android\src\main\java\io\github\edufolly\flutterbluetoothserial\FlutterBluetoothSerialPlugin.java:467: error: cannot find symbol
            requestString = new String[]{Manifest.permission.BLUETOOTH_SCAN,
                                                            ^
  symbol:   variable BLUETOOTH_SCAN
  location: class permission
D:\Flutter\src\.pub-cache\git\flutter_bluetooth_serial-49d778bd52b9c29cb1105190ff05c6dd6f2b8617\android\src\main\java\io\github\edufolly\flutterbluetoothserial\FlutterBluetoothSerialPlugin.java:468: error: cannot find symbol
                    Manifest.permission.BLUETOOTH_CONNECT};
                                       ^
`

rrrovalle, 2022-06-17:
Did you already have tried to add  "android" before the word "manifest" ?

Catalunya69, 2022-06-17:
I have solved it.
In flutter_bluetooth_serial/[android]/build.gradle

Change:
compileSdkVersion 30 -> compileSdkVersion 31

Catalunya69, 2022-06-17:
Hi again, 
i have problem when i connected with bluetooth device, in .toAddress(device.address) function:

PlatformException(connect_error, UID 10396 / PID 3613 lacks permission android.permission.BLUETOOTH

`I/flutter ( 3613): 	at android.bluetooth.IBluetoothSocketManager$Stub$Proxy.connectSocket(IBluetoothSocketManager.java:227)`
`I/flutter ( 3613): 	at android.bluetooth.BluetoothSocket.connect(BluetoothSocket.java:404)`
`I/flutter ( 3613): 	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:57)`
`I/flutter ( 3613): 	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:64)`
`I/flutter ( 3613): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler.lambda$onMethodCall$4$io-github-edufolly-flutterbluetoothserial-FlutterBluetoothSerialPlugin$FlutterBl`

any idea? thanks

mbabuglia, 2022-06-17:
@Catalunya69 , are you trying against a Xiaomi?

Catalunya69, 2022-06-17:
> @Catalunya69 , are you trying against a Xiaomi?

Testing with POCO F3

mbabuglia, 2022-06-17:
> > @Catalunya69 , are you trying against a Xiaomi?
> 
> Testing with POCO F3

Try my answer here: https://stackoverflow.com/a/72370969/666228
On a comment on that question, somebody says it  had the same problem with POCO F3 as well.

mbabuglia, 2022-06-17:
> Ohhhh Thanks!!!
> 
> Solved with: `<uses-permission android:name="android.permission.BLUETOOTH" tools:remove="android:maxSdkVersion" />`

Glad to hear!
Regards.

MohsinP07, 2022-12-29:
> You can use our repo in the meantime, just add this to your pubspec.yaml:
> 
> # flutter_bluetooth_serial: ^0.4.0
> flutter_bluetooth_serial: git: url: https://github.com/pst9354/flutter_bluetooth_serial ref: master
@pst9354  This is not working for me. It would be really helpful if we connect and solve this as I am stuck at this issue since a very long time.

pst9354, 2022-12-29:
> You can use our repo in the meantime, just add this to your pubspec.yaml:
> # flutter_bluetooth_serial: ^0.4.0
> flutter_bluetooth_serial: git: url: https://github.com/pst9354/flutter_bluetooth_serial [...]

So, what is the problem?

MohsinP07, 2022-12-30:
> You can use our repo in the meantime, just add this to your pubspec.yaml:
> 
> # flutter_bluetooth_serial: ^0.4.0
> flutter_bluetooth_serial: git: url: https://github.com/pst9354/flutter_bluetooth_serial ref: master

I used approach as you said here but I am still having problem in connecting to devices with android version 12 and 13.
Are there any changes in manifest file or other that I need to do? 
I have added dependency and version you used as well ie. 0.4.0 
and also in pubspec.yaml I have used this:
flutter_bluetooth_serial:
  git:
  url: https://github.com/pst9354/flutter_bluetooth_serial
  ref: master

pst9354, 2022-12-30:
It's very difficult to help you, if you don't give us more information.
Error logs?

MohsinP07, 2022-12-30:
Only error occurs is that when I try to connect any mobile device with the android version 12 or 13 using Bluetooth it doesn't connect to the device instead it shows the error as :
`cancelDiscovery(): called by: packageName `
 So, I used the solution suggested by you but still I am facing the issue . I just want to know what additional changes I need to do in my project to resolve this issue.

pst9354, 2022-12-30:
I guess Google is your friend, make sure you have the permissions set up correctly.
From our project:
...
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
...

MohsinP07, 2022-12-31:
> I guess Google is your friend, make sure you have the permissions set up correctly. From our project: ...
>
> ```
>     <uses-permission android:name="android.permission.BLUETOOTH" /> [...]

@pst9354 Thanks! After setting these permissions, I am now able to connect the devices. Cheers!!

Zalexei, 2026-11-02:
This is an old thread, but in case someone still needs it: I wrote
bluetooth_serial_next, which works on Android 12+ and requests the new
BLUETOOTH_SCAN/CONNECT permissions itself.
https://pub.dev/packages/bluetooth_serial_next

I've tested it with an ESP32 but not yet with an HC-05 or ELM327.