edufolly/flutter_bluetooth_serial #170: Some Android phones cannot get BLUETOOTH permission while running

fuhongbing134, 2022-05-23:
### Problem summary
Hi, I use this library when I develop a flutter app.  When I use BluetoothConnection.toAddress() to connect a device on XIAOMI, I get a message like this: "lacks permission android.permission.BLUETOOTH". However, I do not meet the same problem on other phones. I met this problem on XIAOMI(Android 12) while the app can run as I want on OPPO(Android 12) and some other phones(Android 10, Android 11) even another XIAOMI(Android 10).

### The permission in AndroidManifest.xml

    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION"/>

    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT"/>
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE"/>

### The request of the permission 

    DeviceInfoPlugin deviceInfo = DeviceInfoPlugin();

    if(Platform.isAndroid) {
      AndroidDeviceInfo androidInfo  = await deviceInfo.androidInfo;
      if(androidInfo.version.sdkInt >= 31) {
        await [Permission.bluetoothConnect, Permission.bluetoothScan , Permission.bluetoothAdvertise, Permission.location].request();
      } else {
        await [Permission.bluetooth, Permission.location].request();
    } else {
      print('ios not yet supported');
    }

### Environment

Flutter assets will be downloaded from https://storage.flutter-io.cn. Make sure you trust this source!
[√] Flutter (Channel stable, 2.10.3, on Microsoft Windows [Version 10.0.19043.1706], locale zh-CN)
    • Flutter version 2.10.3 at F:\flutter_windows\flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 7e9793dee1 (3 months ago), 2022-03-02 11:23:12 -0600
    • Engine revision bd539267b4
    • Dart version 2.16.1
    • DevTools version 2.9.2
    • Pub download mirror https://pub.flutter-io.cn
    • Flutter download mirror https://storage.flutter-io.cn

[√] Android toolchain - develop for Android devices (Android SDK version 31.0.0)
    • Android SDK at C:\Users\PC\AppData\Local\Android\sdk
    • Platform android-31, build-tools 31.0.0
    • Java binary at: C:\Program Files\Android\Android Studio\jre\bin\java
    • Java version OpenJDK Runtime Environment (build 11.0.11+9-b60-7590822)
    • All Android licenses accepted.

[√] Chrome - develop for the web
    • Chrome at C:\Program Files\Google\Chrome\Application\chrome.exe

[!] Visual Studio - develop for Windows (Visual Studio Community 2017 15.9.37)
    • Visual Studio at G:\Microsoft Visual Studio\2017\Community
    • Visual Studio Community 2017 version 15.9.28307.1585
    • Windows 10 SDK version 10.0.20348.0
    X Visual Studio 2019 or later is required.
      Download at https://visualstudio.microsoft.com/downloads/.
      Please install the "Desktop development with C++" workload, including all of its default components

[√] Android Studio (version 2021.1)
    • Android Studio at C:\Program Files\Android\Android Studio
    • Flutter plugin can be installed from:
       https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
       https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.11+9-b60-7590822)

[√] VS Code (version 1.67.0)
    • VS Code at C:\Users\PC\AppData\Local\Programs\Microsoft VS Code
    • Flutter extension version 3.40.0

[√] Connected device (3 available)
    • Windows (desktop) • windows • windows-x64    • Microsoft Windows [Version 10.0.19043.1706]
    • Chrome (web)      • chrome  • web-javascript • Google Chrome 101.0.4951.67
    • Edge (web)        • edge    • web-javascript • Microsoft Edge 101.0.1210.47

[!] HTTP Host Availability
    X HTTP host https://maven.google.com/ is not reachable. Reason: An error occurred while checking the HTTP host:  timeout

```
```

aaronchen96, 2022-10-07:
Hi Fuhongbing, may i know did  u found a way to fix this error?

dungnt1572000, 2023-05-04:
Hi Fuhongbing, this bug happened in Android 12 version because some permission is runtime Permission. you can check some PR relate to Android 12 to resolve it.