edufolly/flutter_bluetooth_serial #172: receiving data continuously

TahaKh99, 2022-06-03:
I have been working on building a flutter app to communicate with a microcontroller where I want to receive data from the microcontroller to my app. The issue which I am facing is that when I send 100 values for example to gether from the other end, I receive the data with some lose and sometimes its divivded in 2 messages. I am not sure what is going on but I will post my code and screen shots of the issues to give a better understanding. 
```
void initState() {
    BluetoothConnection.toAddress(widget.server.address).then((_connection) {
      print('Connected to the device');
      connection = _connection;
      setState(() {
        isConnecting = false;
        isDisconnecting = false;
      });

      connection!.input!.listen(_onDataReceived).onDone(() {
        if (isDisconnecting) {
          print('Disconnecting locally!');
        } else {
          print('Disconnected remotely!');
        }
        if (this.mounted) {
          setState(() {});
        }
      });
    }).catchError((error) {
      print('Cannot connect, exception occured');
      print(error);
    });
  }
```

And here is the function onDataReceived
```
Future<void> _onDataReceived( Uint8List data ) async {
    var fixedList = new List<String>.filled(6, '', growable: false);
    var fixed1 = new List<int>.filled(7, 0, growable: false);
    
    if (data.length >= 7) {
      for (int k = 0; k < 7; k++) {

        fixed1[k] = data[k+1];
        l++;
      }
      if (l > 6) {
        for (int i = 0; i <= 5; i++) {

          fixedList[i] = String.fromCharCode(fixed1[i]);
          fixed1[i]=0;
        }
        l = 0;
        fixed1.clear();
        print("this is data: $data");
        print(fixedList); 
      }
    }
  }
```

this is the results: 
[image] 

The values should be in a form like this: [+/-,0.000] an example would be +0.863
but as you can see its very random 

I tried endless number of ways to solve it but nothing seems to work at all. Please I need help with this matter. thank you all

Syutkin, 2022-06-03:
you can try to split the messages with divider:

```
String _messageBuffer = '';
String _messagePacket = '';

final StreamController<String> _messageController = StreamController.broadcast();

void _onDataReceived(Uint8List data) {
    // Create message if there is '\r\n' sequence
    _messageBuffer += String.fromCharCodes(data);
    while (_messageBuffer.contains('\r\n')) {
      final int index = _messageBuffer.indexOf('\r\n');
      _messagePacket = _messageBuffer.substring(0, index).trim();
      _messageController.add(_messagePacket);
      _messageBuffer = _messageBuffer.substring(index + 2);
    }
  }
```

TahaKh99, 2022-06-03:
Thanks. It wordef perfectly.

TahaKh99, 2022-06-07:
@Syutkin When I tried to convert the string to a double I get an error saying: [ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: FormatException: Invalid double 

`print(double.parse(_messagePacket));`

Syutkin, 2022-06-07:
Please provide more info, 'cos I did'n see any problem with +/- signs
'''
print(double.parse('+0.863'));
print(double.parse('-0.863'));
'''