edufolly/flutter_bluetooth_serial #173: Android bluetooth connection permission for flutter app

TahaKh99, 2022-06-15:
I am using flutter bluetooth serial package on my appliction to do some actions. I was testing my app on a specific android device and it was working completely fine. However, when I tried to use another android device, I started getting errors about not getting the bluetooth permission and all the function in the app which is related to bluetooth is not working and the app is crashing. 
 I have these permissions on my Mainfest file but Im still facing this problem. 

> uses-permission android:name="android.permission.BLUETOOTH" />
> uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
> uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />

The errors are like: 

> getAddress, null, java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for android.content.AttributionSource@267240bd: getAddress

>  Unhandled Exception: PlatformException(error, Need android.permission.BLUETOOTH_CONNECT permission for android.content.AttributionSource@6757bf8c: getName, null, java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for android.content.AttributionSource@6757bf8c: getName

> Caused by: java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for android.content.AttributionSource@bd23a645: AdapterService getBondedDevices

And many more but they all related to bluetooth connection permission. 

Any idea of what to do in my case?

closed 2022-06-16 by TahaKh99

aaronchen96, 2022-10-07:
@TahaKh99 i+Hi, may i know did u found a way to fix this error?

TahaKh99, 2022-10-07:
@aaronchen96 Hi Aaron, 
I fixed it by changing some of the permissions on the package. 

You can use my forked version as this is what I am using on my application and so far it's working perfectly.

https://github.com/TahaKh99/flutter_bluetooth_serial