edufolly/flutter_bluetooth_serial #174: Unhandled Exception: PlatformException(connect_error, read failed, socket might closed or timeout, read ret: -1, java.io.IOException: read failed, socket might closed or timeout, read ret: -1

RRohitM, 2022-06-15:
hello any idea about that?
 
 [ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: PlatformException(connect_error, read failed, socket might closed or timeout, read ret: -1, java.io.IOException: read failed, socket might closed or timeout, read ret: -1
E/flutter (26442): 	at android.bluetooth.BluetoothSocket.readAll(BluetoothSocket.java:970)
E/flutter (26442): 	at android.bluetooth.BluetoothSocket.readInt(BluetoothSocket.java:984)
E/flutter (26442): 	at android.bluetooth.BluetoothSocket.connect(BluetoothSocket.java:536)
E/flutter (26442): 	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:57)
E/flutter (26442): 	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:64)
E/flutter (26442): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler.lambda$onMethodCall$4$FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler(FlutterBluetoothSerialPlugin.java:1007)
E/flutter (26442): 	at io.github.edufolly.flutterbluetoothserial.-$$Lambda$FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler$DZLg1SWj2Aas0tcY7uTlx8xsy4Y.run(Unknown Source:10)
E/flutter (26442): 	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:305)
E/flutter (26442): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
E/flutter (26442): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
E/flutter (26442): 	at java.lang.Thread.run(Thread.java:923)
E/flutter (26442): , null)
E/flutter (26442): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:607:7)
E/flutter (26442): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:177:18)
E/flutter (26442): <asynchronous suspension>
E/flutter (26442): #2      BluetoothConnection.toAddress (package:flutter_bluetooth_serial/BluetoothConnection.dart:61:53)
E/flutter (26442): <asynchronous suspension>
E/flutter (26442): #3      FlutterBluetoothSerial.connectToAddress.<anonymous closure> (package:flutter_bluetooth_serial/FlutterBluetoothSerial.dart:262:30)
E/flutter (26442): <asynchronous suspension>
E/flutter (26442):

jlvas, 2022-06-24:
I'm having the same error

korjaaja, 2022-06-24:
Same error here also.

korjaaja, 2022-06-26:
I just resolved my issue!

It turned out, that for me the problem wasn't in the Android API (android.bluetooth.BluetoothAdapter | android.bluetooth.BluetoothDevice; | android.bluetooth.BluetoothSocket etc..) but in my own dart file..

I had two subsequent calls to **"await BluetoothConnection.toAddress(address)"**, which then resulted in collision at the BluetoothSocket.java, resulting in: **"java.io.IOException: read failed, socket might closed or timeout, read ret: -1"**.

enoearias, 2022-06-29:
D/FlutterBluePlugin( 4757): Connecting to A4:CF:12:72:93:2A (id: 2)
I/flutter ( 4757): Cannot connect, exception occured
I/flutter ( 4757): PlatformException(connect_error, read failed, socket might closed or timeout, read ret: -1, java.io.IOException: read failed, socket might closed or timeout, read ret: -1
I/flutter ( 4757): 	at android.bluetooth.BluetoothSocket.readAll(Unknown Source:69)
I/flutter ( 4757): 	at android.bluetooth.BluetoothSocket.readInt(Unknown Source:3)
I/flutter ( 4757): 	at android.bluetooth.BluetoothSocket.connect(Unknown Source:118)
I/flutter ( 4757): 	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:57)
I/flutter ( 4757): 	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:64)
I/flutter ( 4757): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler.lambda$onMethodCall$4$FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler(FlutterBluetoothSerialPlugin.java:1007)
I/flutter ( 4757): 	at io.github.edufolly.flutterbluetoothserial.-$$Lambda$FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler$DZLg1SWj2Aas0tcY7uTlx8xsy4Y.run(Unknown Source:10)
I/flutter ( 4757): 	at android.os.AsyncTask$Seri
I/flutter ( 4757): Connect -> selected A4:CF:12:72:93:2A
D/FlutterBluePlugin( 4757): Connecting to A4:CF:12:72:93:2A (id: 3)

abdulknarkz, 2022-07-18:
I/flutter ( 8373): 14:C6:97:4B:AC:02
D/FlutterBluePlugin( 8373): Connecting to 14:C6:97:4B:AC:02 (id: 7)
W/BluetoothAdapter( 8373): getBluetoothService() called with no BluetoothManagerCallback
I/flutter ( 8373): Cannot connect, exception occurred
I/flutter ( 8373): PlatformException(connect_error, read failed, socket might closed or timeout, read ret: -1, java.io.IOException: read failed, socket might closed or timeout, read ret: -1
I/flutter ( 8373): 	at android.bluetooth.BluetoothSocket.readAll(BluetoothSocket.java:758)
I/flutter ( 8373): 	at android.bluetooth.BluetoothSocket.readInt(BluetoothSocket.java:772)
I/flutter ( 8373): 	at android.bluetooth.BluetoothSocket.connect(BluetoothSocket.java:404)
I/flutter ( 8373): 	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:57)
I/flutter ( 8373): 	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:64)
I/flutter ( 8373): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler.lambda$onMethodCall$4$io-github-edufolly-flutterbluetoothserial-FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler(FlutterBluetoothSerialPlugin.java:1007)
I/flutter ( 8373): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler$$ExternalSyntheticLambda2.r

KarolNiemiec, 2022-09-20:
same problem

guigzp, 2022-09-26:
Any updates on this? I'm facing the same problem

manandharsudhir, 2022-11-16:
do we have any solutions to this?

yavuzfk, 2023-04-17:
If the matter is data tranfering without internet connection, you can use wifi-direct packages instead of bluetooth. I solved with this [package](https://pub.dev/packages/flutter_p2p_connection) .

SebiAndrei223, 2023-06-16:
Same error here, any updates on a fix?

ahmeedev, 2024-04-06:
I am facing this same issue ! 😭

safu2011, 2024-05-24:
Same issue

mnasirbaloch, 2024-07-12:
any solution?

mnasirbaloch, 2024-07-12:
> I am facing this same issue ! 😭

did you find any solution?

ClickMakers, 2024-08-18:
> > I am facing this same issue ! 😭
> 
> did you find any solution?

Same issue any solution?