edufolly/flutter_bluetooth_serial #177: Incomplete incoming messages using Arduino NANO module HC-06

Rizardomon, 2022-07-04:
Hello @edufolly

I am developing an application using a library to control arduino boards via bluetooth.
We started here using an ESP-32 board and developed everything based on it and everything is working perfectly, but at the end of the development we also need to use an arduino NANO board with a HC-06 module to have the bluetooth and with this board the messages are coming cut off in the flutter application, sometimes cutting the beginning of the message and other times at the end.

The connection method is very simple, it just checks if there is already a device saved previously and if it is not connected to try to connect and when it loses the connection it tries to reconnect.

And the method that receives the data is just printing on the screen what is being received from the arduino.

[image]

[image]

### Image with the standard message that arrives in the ESP-32:
<div>[image]</div>

### Images from HC-06 NANO board with the messages cut
<div>
[image]
[image]
[image]
</div>

We thought it could be a problem with the HC-06 or NANO board, but when it sends and receives from the ESP-32 the messages arrive and are sent correctly. Even using an external serial app the messages are correct too.
[image]

From our own experience we think it involves speed, because the arduino code waits until the whole message arrives at the device and we don't understand if this is possible to do with this library.

iqfareez, 2023-01-14:
Same issue. The messages received were either incomplete or empty. We are using HC-05 bluetooth module with Arduino UNO.

AlecDr, 2023-03-23:
Hey @Rizardomon

I think you should check the library example to tackle this issue, specifically the chat page code, in the input listener there is some magic being done to get the messages complete, basically you should define a message buffer to get the complete message when the device signals the "last" message.