edufolly/flutter_bluetooth_serial #181: Failed build example app in release mode

iqfareez, 2022-09-24:
## Steps to reproduce

1. Clone project
2. `cd example`
3. `flutter run --release`

## Expected behavior

App run normally as it would on debug

## Actual behavior

Error when building the application.

```
Running "flutter pub get" in example...                            927ms
Launching lib\main.dart on YES Altitude 4 in release mode...

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:lintVitalRelease'.
> Could not resolve all artifacts for configuration ':app:debugAndroidTestRuntimeClasspath'.
   > Could not resolve androidx.test:runner:1.4.0.
     Required by:
         project :app
         project :app > androidx.test.espresso:espresso-core:3.4.0
      > Cannot find a version of 'androidx.test:runner' that satisfies the version constraints:
           Dependency path 'android:app:unspecified' --> 'androidx.test:runner:1.4.0'
           Constraint path 'android:app:unspecified' --> 'androidx.test:runner:{strictly 1.2.0}' because of the following reason: debugRuntimeClasspath uses version 1.2.0
           Dependency path 'android:app:unspecified' --> 'androidx.test.espresso:espresso-core:3.4.0' --> 'androidx.test:runner:1.4.0'

   > Could not resolve androidx.test.espresso:espresso-core:3.4.0.
     Required by:
         project :app
      > Cannot find a version of 'androidx.test.espresso:espresso-core' that satisfies the version constraints:
           Dependency path 'android:app:unspecified' --> 'androidx.test.espresso:espresso-core:3.4.0'
           Constraint path 'android:app:unspecified' --> 'androidx.test.espresso:espresso-core:{strictly 3.2.0}' because of the following reason: debugRuntimeClasspath uses version 3.2.0

   > Could not resolve androidx.test:runner:{strictly 1.2.0}.
     Required by:
         project :app
      > Cannot find a version of 'androidx.test:runner' that satisfies the version constraints:
           Dependency path 'android:app:unspecified' --> 'androidx.test:runner:1.4.0'
           Constraint path 'android:app:unspecified' --> 'androidx.test:runner:{strictly 1.2.0}' because of the following reason: debugRuntimeClasspath uses version 1.2.0
           Dependency path 'android:app:unspecified' --> 'androidx.test.espresso:espresso-core:3.4.0' --> 'androidx.test:runner:1.4.0'

   > Could not resolve androidx.test.espresso:espresso-core:{strictly 3.2.0}.
     Required by:
         project :app
      > Cannot find a version of 'androidx.test.espresso:espresso-core' that satisfies the version constraints:
           Dependency path 'android:app:unspecified' --> 'androidx.test.espresso:espresso-core:3.4.0'
           Constraint path 'android:app:unspecified' --> 'androidx.test.espresso:espresso-core:{strictly 3.2.0}' because of the following reason: debugRuntimeClasspath uses version 3.2.0

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights. 

* Get more help at https://help.gradle.org

BUILD FAILED in 2s
Running Gradle task 'assembleRelease'...                            3.9s
Exception: Gradle task assembleRelease failed with exit code 1
#0      throwToolExit (package:flutter_tools/src/base/common.dart:10:3)
#1      AndroidGradleBuilder.buildGradleApp (package:flutter_tools/src/android/gradle.dart:408:9)
<asynchronous suspension>
#2      AndroidGradleBuilder.buildApk (package:flutter_tools/src/android/gradle.dart:184:5)
<asynchronous suspension>
#3      AndroidDevice.startApp (package:flutter_tools/src/android/android_device.dart:575:7)
<asynchronous suspension>
#4      FlutterDevice.runCold (package:flutter_tools/src/resident_runner.dart:504:33)
<asynchronous suspension>
#5      ColdRunner.run (package:flutter_tools/src/run_cold.dart:57:28)
<asynchronous suspension>
#6      RunCommand.runCommand (package:flutter_tools/src/commands/run.dart:706:27)
<asynchronous suspension>
#7      FlutterCommand.run.<anonymous closure> (package:flutter_tools/src/runner/flutter_command.dart:1209:27)
<asynchronous suspension>
#8      AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
<asynchronous suspension>
#9      CommandRunner.runCommand (package:args/command_runner.dart:209:13)
<asynchronous suspension>
#10     FlutterCommandRunner.runCommand.<anonymous closure> (package:flutter_tools/src/runner/flutter_command_runner.dart:281:9)
<asynchronous suspension>
#11     AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
<asynchronous suspension>
#12     FlutterCommandRunner.runCommand (package:flutter_tools/src/runner/flutter_command_runner.dart:229:5)
<asynchronous suspension>
#13     run.<anonymous closure>.<anonymous closure> (package:flutter_tools/runner.dart:62:9)
<asynchronous suspension>
#14     AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
<asynchronous suspension>
#15     main (package:flutter_tools/executable.dart:91:3)
<asynchronous suspension>
```

imammashur, 2022-10-12:
got same issue. actually I was looking for these classing bluetooth instead of using BLE now.

ganxiaozhe, 2022-11-20:
modify `/android/app/build.gradle`
```
dependencies {
    testImplementation 'junit:junit:4.13.2'
    androidTestImplementation 'androidx.test:runner:1.4.0'
    androidTestImplementation 'androidx.test.espresso:espresso-core:3.4.0'
}
```
to
```
dependencies {
    testImplementation 'junit:junit:4.13.2'
    androidTestImplementation 'androidx.test:runner:1.2.0'
    androidTestImplementation 'androidx.test.espresso:espresso-core:3.2.0'
}
```