edufolly/flutter_bluetooth_serial #184: Cannot connect to another android device.

AlphaRosh, 2022-12-03:
Hello, i am trying to connect and communicate between two android devices.

I am using a Vivo v20 pro and samsung galaxy tab (SMT515) for testing. Both have the demo serial app installed.

To Reproduce: 
1) Open the app on both phone and tab and turned on Bluetooth and location for both of them. 
2) Making one device as discoverable and searching for the device from the other. (I am able to get the device and also pair it.)
3) Go to "Connect to paired device to chat" with both phone and tab, select the device to chat.

On connecting to the device from vs , i get the following exception on trying to chat with the connected device.

[image]

 > Exception has occurred.
PlatformException (PlatformException(connect_error, read failed, socket might closed or timeout, read ret: -1, java.io.IOException: read failed, socket might closed or timeout, read ret: -1
	at android.bluetooth.BluetoothSocket.readAll(BluetoothSocket.java:970)
	at android.bluetooth.BluetoothSocket.readInt(BluetoothSocket.java:984)
	at android.bluetooth.BluetoothSocket.connect(BluetoothSocket.java:536)
	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:57)
	at io.github.edufolly.flutterbluetoothserial.BluetoothConnection.connect(BluetoothConnection.java:64)
	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler.lambda$onMethodCall$4$FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler(FlutterBluetoothSerialPlugin.java:1007)
	at io.github.edufolly.flutterbluetoothserial.-$$Lambda$FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler$DZLg1SWj2Aas0tcY7uTlx8xsy4Y.run(Unknown Source:10)
	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:305)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
	at java.lang.Thread.run(Thread.java:923)
, null))

Thanks

dev9828, 2023-01-03:
hey? did you solve the issue? i cant connect two phone with same aplication.

SrinivasaMurthy91, 2023-05-12:
anyone solve the issue? i am unable to connect two mobiles

udit6023, 2023-06-11:
BluetoothConnection newConnection =
        await BluetoothConnection.toAddress(device.address);
i have passed a raw device's address in the above code still it is not able to connect to that device
were you able to connect to your desired device?

SrinivasaMurthy91, 2023-06-26:
I am unable to connect to desired device

zamirszn, 2024-02-20:
I'm facing this too has anyone been able to figure it out