edufolly/flutter_bluetooth_serial #187: I add java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT but it doesn't work

FranciscoPV94, 2022-12-20:
### Problem summary

Apparently the problem is due to the new version of Android 12, which needs bluetooth access permissions.
I am testing on a moto g50 with android 12, I tried on a moto g70 with android 11 and everything works fine.

in android docs find this [https://developer.android.com/guide/topics/connectivity/bluetooth/permissions#declare-android12-or-higher](https://developer.android.com/guide/topics/connectivity/bluetooth/permissions#declare-android12-or-higher)

### Steps to reproduce

1. E/MethodChannel#flutter_bluetooth_serial/methods( 4896): Failed to handle method call
E/MethodChannel#flutter_bluetooth_serial/methods( 4896): java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for android.content.AttributionSource@8702d63b: getName

I add java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT but it doesn't work

### Environment

flutter --version 
Flutter 2.2.3 • channel stable • https://github.com/flutter/flutter.git
Framework • revision f4abaa0735 (1 year, 6 months ago) • 2021-07-01 12:46:11 -0700
Engine • revision 241c87ad80
Tools • Dart 2.13.4

Lauritz-Tieste, 2022-12-28:
You need to uninstall the app completely from your device and reinstall it. Make sure, that you have granted all the permissions. This worked at my case (with the same error).

MazenCoder, 2023-01-17:
Same issue, also not working on the release mode

tomc128, 2023-04-21:
Have you explicitly requested the permission from within the app? From that Android developer website link:

> The BLUETOOTH_ADVERTISE, BLUETOOTH_CONNECT, and BLUETOOTH_SCAN permissions are [runtime permissions](https://developer.android.com/guide/topics/permissions/overview#runtime). Therefore, you must explicitly [request user approval](https://developer.android.com/training/permissions/requesting) in your app before you can look for Bluetooth devices, make a device discoverable to other devices, or communicate with already-paired Bluetooth devices. When your app requests at least one of these permissions, the system prompts the user to allow your app to access Nearby devices, as shown in figure 1.

To do this, use the [permission_handler](https://pub.dev/packages/permission_handler) plugin.