edufolly/flutter_bluetooth_serial #197: App launches but immediately freezes and crashes

ShadowStacker, 2023-04-03:
Trying to run the example from 0.4.0 on my Pixel 5a running whatever the latest flavor of android is available for it and I get this Debug Output:

`Launching lib\main.dart on Pixel 5a in debug mode...
main.dart:1
Warning: Mapping new ns http://schemas.android.com/repository/android/common/02 to old ns http://schemas.android.com/repository/android/common/01
Warning: Mapping new ns http://schemas.android.com/repository/android/generic/02 to old ns http://schemas.android.com/repository/android/generic/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/addon2/02 to old ns http://schemas.android.com/sdk/android/repo/addon2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/addon2/03 to old ns http://schemas.android.com/sdk/android/repo/addon2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/repository2/02 to old ns http://schemas.android.com/sdk/android/repo/repository2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/repository2/03 to old ns http://schemas.android.com/sdk/android/repo/repository2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/sys-img2/03 to old ns http://schemas.android.com/sdk/android/repo/sys-img2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/sys-img2/02 to old ns http://schemas.android.com/sdk/android/repo/sys-img2/01
Warning: Mapping new ns http://schemas.android.com/repository/android/common/02 to old ns http://schemas.android.com/repository/android/common/01
Warning: Mapping new ns http://schemas.android.com/repository/android/generic/02 to old ns http://schemas.android.com/repository/android/generic/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/addon2/02 to old ns http://schemas.android.com/sdk/android/repo/addon2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/addon2/03 to old ns http://schemas.android.com/sdk/android/repo/addon2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/repository2/02 to old ns http://schemas.android.com/sdk/android/repo/repository2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/repository2/03 to old ns http://schemas.android.com/sdk/android/repo/repository2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/sys-img2/03 to old ns http://schemas.android.com/sdk/android/repo/sys-img2/01
Warning: Mapping new ns http://schemas.android.com/sdk/android/repo/sys-img2/02 to old ns http://schemas.android.com/sdk/android/repo/sys-img2/01
√  Built build\app\outputs\flutter-apk\app-debug.apk.
V/FlutterBluetoothSerial(23048): Attached to engine
I/thserialexample(23048): Compiler allocated 4413KB to compile void android.view.ViewRootImpl.performTraversals()
Connecting to VM Service at ws://127.0.0.1:64348/Gcsxj9xW4x4=/ws
E/MethodChannel#flutter_bluetooth_serial/methods(23048): Failed to handle method call
E/MethodChannel#flutter_bluetooth_serial/methods(23048): java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for android.content.AttributionSource@5da1dd98: getName
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at android.os.Parcel.createExceptionOrNull(Parcel.java:3011)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at android.os.Parcel.createException(Parcel.java:2995)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at android.os.Parcel.readException(Parcel.java:2978)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at android.os.Parcel.readException(Parcel.java:2920)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at android.bluetooth.IBluetoothManager$Stub$Proxy.getName(IBluetoothManager.java:555)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at android.bluetooth.BluetoothAdapter.getName(BluetoothAdapter.java:1552)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler.onMethodCall(FlutterBluetoothSerialPlugin.java:727)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$DartMessenger(DartMessenger.java:322)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at io.flutter.embedding.engine.dart.-$$Lambda$DartMessenger$2j2MERcK825A5j1fv5sZ7xB2Iuo.run(Unknown Source:12)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at android.os.Handler.handleCallback(Handler.java:942)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at android.os.Looper.loopOnce(Looper.java:201)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at android.os.Looper.loop(Looper.java:288)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at android.app.ActivityThread.main(ActivityThread.java:7884)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at java.lang.reflect.Method.invoke(Native Method)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): Caused by: android.os.RemoteException: Remote stack trace:
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at com.android.server.bluetooth.BluetoothManagerService.checkPermissionForDataDelivery(BluetoothManagerService.java:3248)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at com.android.server.bluetooth.BluetoothManagerService.checkConnectPermissionForDataDelivery(BluetoothManagerService.java:3266)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at com.android.server.bluetooth.BluetoothManagerService.getName(BluetoothManagerService.java:1942)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at android.bluetooth.IBluetoothManager$Stub.onTransact(IBluetoothManager.java:258)
E/MethodChannel#flutter_bluetooth_serial/methods(23048): 	at android.os.Binder.execTransactInternal(Binder.java:1280)
E/MethodChannel#flutter_bluetooth_serial/methods(23048):
W/FlutterBluePlugin(23048): Local Bluetooth MAC address is hidden by system, trying other options...
D/FlutterBluePlugin(23048): Trying to obtain address using Settings Secure bank
D/FlutterBluePlugin(23048): Obtaining address using Settings Secure bank failed
D/FlutterBluePlugin(23048): Trying to obtain address using reflection against internal Android code
W/thserialexample(23048): Accessing hidden field Landroid/bluetooth/BluetoothAdapter;->mService:Landroid/bluetooth/IBluetooth; (unsupported, reflection, allowed)
W/thserialexample(23048): Accessing hidden method Landroid/bluetooth/IBluetooth$Stub$Proxy;->getAddress()Ljava/lang/String; (unsupported, reflection, allowed)
D/FlutterBluePlugin(23048): Obtaining address using reflection against internal Android code failed
D/FlutterBluePlugin(23048): Trying to look up address by network interfaces - might be invalid on some devices
D/CompatibilityChangeReporter(23048): Compat change id reported: 170188668; UID 10347; state: ENABLED
W/FlutterBluePlugin(23048): Looking for address by network interfaces failed
E/SurfaceSyncer(23048): Failed to find sync for id=0
W/Parcel  (23048): Expecting binder but got null!
E/OpenGLRenderer(23048): Unable to match the desired swap behavior.
E/SurfaceSyncer(23048): Failed to find sync for id=0
E/SurfaceSyncer(23048): Failed to find sync for id=1
Lost connection to device.
Exited (sigterm)`

AlecDr, 2023-04-05:
Hi @ShadowStacker,

This is probably a permission issue, in the log you provided there is a line where it says "java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT".

Try to define the needed android permissions in your android/app/main/AndroidManifest.xml before the application tag (if that doesnt work, try defining the following in the debug manifest too);

``` 
    <manifest ...

    <!-- Permissions options for the `nearby devices` group -->
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />

    <application ...
```

ShadowStacker, 2023-04-06:
Thanks for the reply! I had already added those lines to my AndroidManifest.xml file, I don't have a debug manifest. I created a new folder in src and named it "debug" and copied the manifest file into it but I have the same error log and same behavior of the app freezing.

ShadowStacker, 2023-04-07:
AFAIK I am running the latest version of Flutter in VS Code. 
Also sorry I am very new to this (I have lots of expierence with arduino IDE but I am trying my hand at making a simple app using BLE to control an ESP32) and I dont understand how to run a command in the android folder. Can you elaborate. Thank you kindly!

AlecDr, 2023-04-07:
Generally when something goes wrong in my projects after a new package/library installation or configuration change the first thing i do is to run the gradlew utility that resides in the android folder of the project with the clean option, generally you cd into the android folder of your project and run the "gradlew clean" command, that will try to clean some old configuration of your project, always good to try.

Also, you will be using the ESP32 with bluetooth low energy or serial to communicate with the app? This package only works with serial communication if i'm not wrong, so if you're planning to use the low energy counterpart i would use something like  [this package](https://github.com/PhilipsHue/flutter_reactive_ble) that works for BLE.

Another thing you can try is to run the [example project](https://github.com/edufolly/flutter_bluetooth_serial/tree/master/example) of this package to see if there's something wrong with your project.

ShadowStacker, 2023-04-07:
Sorry I meant to say serial instead of BLE, but I may switch over to BLE because this is a hassle. I still dont understand how to run commands within VS code, other than ctrl-shift-p.

AlecDr, 2023-04-07:
In VSCode you can run commands through the [integrated terminal](https://code.visualstudio.com/docs/terminal/basics), you can also run the commands through your native OS terminal, cmd in windows or the terminal in linux.

Honestly i think that the first thing you should do is to run the example project i said earlier, just to make sure that there is nothing wrong with your setup.

I'm also currently using this package to develop a app that uses serial communication from a buetooth device, but i'm using flutter 3.0, maybe the current version of flutter has some kind of problem with this package since it's not updated in a while.

ShadowStacker, 2023-04-07:
The example project is what I am trying to run, in fact I am using a slightly updated version hosted here: https://github.com/pst9354/flutter_bluetooth_serial

AlecDr, 2023-04-08:
In that case the project is created by you or you downloaded the source code of this package?

If you want to run the command for the example project of this package you should run the gradlew command inside the android folder from the example folder, something like this:
```
cd D:\src\flutter_bluetooth_serial\example\android\ && gradlew clean
```

Now, if the project you created happens to have the same name from this package i would run the ``` flutter doctor ``` command, just to see if theres nothing wrong with your setup.