edufolly/flutter_bluetooth_serial #200: App crashes on Android 12 or after!

savitore, 2023-07-15:
### Problem summary

When we add the Bluetooth connectivity to the app and visit the screen that contains the connection, the app crashes if the android mobile has Android 12 or after. The connection works fine for Android 11 or before.
There might be some permissions problem, but I've tried adding every Bluetooth permission, and it still doesn't work. 

### Steps to reproduce

1. 
2. 
3. 
4. 

```
```

Fouad-Asil, 2023-08-18:
try to add [ACCESS_FINE_LOCATION](https://developer.android.com/reference/android/Manifest.permission#ACCESS_FINE_LOCATION)

savitore, 2023-08-18:
added that, didn't work:)

dotupNET, 2023-08-30:
Try this

`
   <uses-permission android:name="android.permission.BLUETOOTH" />
   <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
   <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
   <uses-permission android:name="android.permission.BLUETOOTH_SCAN"
      android:usesPermissionFlags="neverForLocation" />
   <uses-permission-sdk-23 android:name="android.permission.ACCESS_COARSE_LOCATION"
      android:maxSdkVersion="30" />
   <uses-permission-sdk-23 android:name="android.permission.ACCESS_FINE_LOCATION"
      android:maxSdkVersion="30" />
`

GabLast, 2023-09-08:
Having the same Issue here.

When the permissions page pop up and they are granted, the app crashes. However, when the permissions are denied, everything's fine.

Tested with the current manifest:
 
    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN"/>
    
    <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE"/>
    
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT"/>
    
    <uses-feature android:name="android.hardware.bluetooth" android:required="false"/>
    <uses-feature android:name="android.hardware.bluetooth_le" android:required="false"/>

    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>

xtensa, 2023-10-09:
Bump. Any update on this issue? This is quite critical issue, as it is drawing plugin unusable on Android 12+.