edufolly/flutter_bluetooth_serial #207: Unable to connect to my bluetooth module which is sends string of data

zubair999, 2023-10-20:
I have bluetooth module which is sending some data string continously, I have install Serial Bluetooth Termial
https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=en&gl=US
this app from play store and I am getting the data when device is connected,

So how to stream data from the bluetooth in flutter_blue_plus.

I already running example app but not getting anywhere, device is connecting but how to get the data stream

zamirszn, 2024-03-11:
> I have bluetooth module which is sending some data string continously, I have install Serial Bluetooth Termial [...]

hello were you able to do this ?? I'm also trying to connect to a BT module

zubair999, 2024-03-15:
> hello were you able to do this ?? I'm also trying to connect to a BT module

Yes

zamirszn, 2024-03-15:
> Yes

can you help with a demo code please , im trying to connect to a raspberry pi using this package but the example code doesnt work

zubair999, 2024-03-16:
> can you help with a demo code please , im trying to connect to a raspberry pi using this package but the example code doesnt work

I suggest you to use flutter blue plus