edufolly/flutter_bluetooth_serial #211: Having Issue in Connecting with Android Versions 12 or more

haisamk18, 2024-01-20:
This is my error:
Cannot connect, exception occurred: PlatformException(connect_error, Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@2d842ac7: AdapterService cancelDiscovery, java.lang.SecurityException: Need android.permission.BLUETOOTH_SCAN permission for android.content.AttributionSource@2d842ac7: AdapterService cancelDiscovery

My AndroidManifest.xml code :

<manifest xmlns:android="http://schemas.android.com/apk/res/android">
    <uses-permission android:name="android.permission.BLUETOOTH"
        android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"
        android:maxSdkVersion="30" />

    
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />

    
    <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />

    
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />

    
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <application
        android:label="nw"
        android:name="${applicationName}"
        android:icon="@mipmap/ic_launcher">
        <activity
            android:name=".MainActivity"
            android:exported="true"
            android:launchMode="singleTop"
            android:theme="@style/LaunchTheme"
            android:configChanges="orientation|keyboardHidden|keyboard|screenSize|smallestScreenSize|locale|layoutDirection|fontScale|screenLayout|density|uiMode"
            android:hardwareAccelerated="true"
            android:windowSoftInputMode="adjustResize">
            
            <meta-data
              android:name="io.flutter.embedding.android.NormalTheme"
              android:resource="@style/NormalTheme"
              />
            <intent-filter>
                <action android:name="android.intent.action.MAIN"/>
                <category android:name="android.intent.category.LAUNCHER"/>
            </intent-filter>
        </activity>
        
        <meta-data
            android:name="flutterEmbedding"
            android:value="2" />
    </application>
</manifest>

Help with the issue although I am not using any Scan logic .

haisamk18, 2024-01-20:
The error is consistently coming but in other Android version it is working fine

fernandomorse, 2024-02-04:
I have the same problem. Any solution to fix. Does not work on SDKs above 30.

zamirszn, 2024-02-19:
also have the same issue , I have android 13 , I added the permissions and gave access from the device settings to get it to work

haisamk18, 2024-02-20:
It is working now.

fernandomorse, 2024-02-21:
@haisamk18 
I still have the same problem, I'm using:

pubspec.yam

dependencies:
    flutter_bluetooth_serial: ^0.4.0

AndroidManifest.xml
    <uses-permission android:name="android.permission.INTERNET"/>
    <uses-permission android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN"/>
   <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE"/>
   <uses-permission android:name="android.permission.BLUETOOTH_CONNECT"/>
   <uses-feature android:name="android.hardware.bluetooth" android:required="false"/>
   <uses-feature android:name="android.hardware.bluetooth_le" android:required="false"/>
     <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
     <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>

app\build.gradle
  compileSdkVersion 31

What should I modify to make it work??

haisamk18, 2024-02-21:
Unfortunately after adding all the permissions in manifest file it won’t
work.But you can manually give permissions to the app on your phone and
them it will work.

fernandomorse, 2024-02-21:
@haisamk18 
Correct manually it works, but I believe they should correct it in the next versions, thank you

zamirszn, 2024-02-21:
> if you ask for permission (nearby device and location) at run time it should work ?

Lokesh-Enata, 2024-02-23:
i was having the same issue don't know whether it has been solved or not but before that i am getting new issue whenever app launches it ask for location permission whenever i click on allow , only for this time or deny button the app crashes and it is only happening for android 13

zamirszn, 2024-02-23:
> i was having the same issue don't know whether it has been solved or not but before that i am getting new issue whenever app launches it ask for location permission whenever i click on allow , only for this time or deny button the app crashes and it is only happening for android 13
> 

I notice this if you don't grant nearby devices permission , try asking for nearby devices permission at run time with permission handler

Lokesh-Enata, 2024-02-23:
<manifest xmlns:android="http://schemas.android.com/apk/res/android">
<uses-permission android:name="android.permission.BLUETOOTH"/>
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
<uses-permission android:name="android.permission.BLUETOOTH_SCAN"/>
    <application
        android:label="demo_ble_project1"
        android:name="${applicationName}"
        android:icon="@mipmap/ic_launcher">
        <activity
            android:name=".MainActivity"
            android:exported="true"
            android:launchMode="singleTop"
            android:theme="@style/LaunchTheme"
            android:configChanges="orientation|keyboardHidden|keyboard|screenSize|smallestScreenSize|locale|layoutDirection|fontScale|screenLayout|density|uiMode"
            android:hardwareAccelerated="true"
            android:windowSoftInputMode="adjustResize">
            
            <meta-data
              android:name="io.flutter.embedding.android.NormalTheme"
              android:resource="@style/NormalTheme"
              />
            <intent-filter>
                <action android:name="android.intent.action.MAIN"/>
                <category android:name="android.intent.category.LAUNCHER"/>
            </intent-filter>
        </activity>
        
        <meta-data
            android:name="flutterEmbedding"
            android:value="2" />
    </application>
    
    <queries>
        <intent>
            <action android:name="android.intent.action.PROCESS_TEXT"/>
            <data android:mimeType="text/plain"/>
        </intent>
    </queries>
</manifest>

this is my android manifest file i have already given permission and even used permission handler package but nothing works this issue is only appearing on Android 13

Lokesh-Enata, 2024-02-23:
> I notice this if you don't grant nearby devices permission , try asking for nearby devices permission at run time with permission handler

<manifest xmlns:android="http://schemas.android.com/apk/res/android">
<uses-permission android:name="android.permission.BLUETOOTH"/>
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
<uses-permission android:name="android.permission.BLUETOOTH_SCAN"/>
    <application
        android:label="demo_ble_project1"
        android:name="${applicationName}"
        android:icon="@mipmap/ic_launcher">
        <activity
            android:name=".MainActivity"
            android:exported="true"
            android:launchMode="singleTop"
            android:theme="@style/LaunchTheme"
            android:configChanges="orientation|keyboardHidden|keyboard|screenSize|smallestScreenSize|locale|layoutDirection|fontScale|screenLayout|density|uiMode"
            android:hardwareAccelerated="true"
            android:windowSoftInputMode="adjustResize">
            
            <meta-data
              android:name="io.flutter.embedding.android.NormalTheme"
              android:resource="@style/NormalTheme"
              />
            <intent-filter>
                <action android:name="android.intent.action.MAIN"/>
                <category android:name="android.intent.category.LAUNCHER"/>
            </intent-filter>
        </activity>
        
        <meta-data
            android:name="flutterEmbedding"
            android:value="2" />
    </application>
    
    <queries>
        <intent>
            <action android:name="android.intent.action.PROCESS_TEXT"/>
            <data android:mimeType="text/plain"/>
        </intent>
    </queries>
</manifest>
this is my android manifest file i have given all the permission and even used permission handler package but nothing works this issue is only appearing on Android 13 device , if u have any solution for this it would be a great help

Lokesh-Enata, 2024-02-23:
i think the flutter bluetooth serial package does not supports android 13

haisamk18, 2024-02-23:
No ,It does. The thing is for some devices it works fine and for some it
gives error even after giving all permission in manifest file but when
giving the permission of nearby devices from phone it works fine.

Lokesh-Enata, 2024-02-23:
yes may be you are right but my app first ask location even if i allow it crashes i m not getting why
even i have turn on nearby share on

Christyansoft, 2024-02-28:
@Lokesh-Enata  same problem here, did you find a solution ?

Lokesh-Enata, 2024-02-29:
> @Lokesh-Enata same problem here, did you find a solution ?

yess , i used flutter bluetooth serial ble package that solved my problem.

zamirszn, 2024-02-29:
> > @Lokesh-Enata same problem here, did you find a solution ?
> 
> yess , i used flutter bluetooth serial ble package that solved my problem.
> 

hello , can you help with a link to the package please

Lokesh-Enata, 2024-02-29:
> > > @Lokesh-Enata same problem here, did you find a solution ?
> > 
> > 
> > yess , i used flutter bluetooth serial ble package that solved my problem.
> 
> hello , can you help with a link to the package please

sure, 'https://pub.dev/packages/flutter_bluetooth_serial_ble'  this is the link for the package.

zamirszn, 2024-02-29:
> > > > @Lokesh-Enata same problem here, did you find a solution ?
> > > 
> > > 
> > > yess , i used flutter bluetooth serial ble package that solved my problem.
> > 
> > hello , can you help with a link to the package please
> 
> sure, 'https://pub.dev/packages/flutter_bluetooth_serial_ble'  this is the link for the package.

thanks alot

Christyansoft, 2024-02-29:
unfutnatelly the flutter [flutter_bluettoth_serial_ble](https://pub.dev/packages/flutter_bluetooth_serial_ble/install) has the same problem. When request location at the first time, crashs the app.

zamirszn, 2024-02-29:
> Unfortunately the flutter [flutter_bluettoth_serial_ble](https://pub.dev/packages/flutter_bluetooth_serial_ble/install) has the same problem. When request location at the first time in android 12 or later,  crashs the app. there are another similar packages that do the same, but the owners sounds like do not approve the pr's 🙃

I'm running a build and looking for a fix , I'll let you know how it goes

zamirszn, 2024-02-29:
> Unfortunately the flutter [flutter_bluettoth_serial_ble](https://pub.dev/packages/flutter_bluetooth_serial_ble/install) has the same problem. When request location at the first time in android 12 or [...]

ive been able to fix this 

you can add 

```
 <uses-permission android:name="android.permission.INTERNET"/>
 <uses-permission android:name="android.permission.BLUETOOTH" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
<uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
```

 in your manifest not all of them are required? 

then before anything else request for the bluetooth Connect permission

` Permission.bluetoothConnect.request()`

you can use the permission_handler package then initialize the bluetooth method , below is the full code 

```
import 'dart:async';

import 'package:flutter/material.dart';
import 'package:flutter_bluetooth_serial_ble/flutter_bluetooth_serial_ble.dart';
import 'package:permission_handler/permission_handler.dart';
import 'package:scoped_model/scoped_model.dart';

import './BackgroundCollectedPage.dart';
import './BackgroundCollectingTask.dart';
import './ChatPage.dart';
import './DiscoveryPage.dart';
import './SelectBondedDevicePage.dart';

// import './helpers/LineChart.dart';

class MainPage extends StatefulWidget {
  @override
  _MainPage createState() => new _MainPage();
}

class _MainPage extends State<MainPage> {
  BluetoothState _bluetoothState = BluetoothState.UNKNOWN;

  String _address = "...";
  String _name = "...";

  Timer? _discoverableTimeoutTimer;
  int _discoverableTimeoutSecondsLeft = 0;

  BackgroundCollectingTask? _collectingTask;

  bool _autoAcceptPairingRequests = false;

  @override
  void initState() {
    // Get current state
    super.initState();
  }

  void initBT() {
    FlutterBluetoothSerial.instance.state.then((state) {
      setState(() {
        _bluetoothState = state;
      });
    });
[... 326 lines cut ...]
      _collectingTask?.cancel();
      showDialog(
        context: context,
        builder: (BuildContext context) {
          return AlertDialog(
            title: const Text('Error occured while connecting'),
            content: Text("${ex.toString()}"),
            actions: <Widget>[
              new TextButton(
                child: new Text("Close"),
                onPressed: () {
                  Navigator.of(context).pop();
                },
              ),
            ],
          );
        },
      );
    }
  }
}

```

and lastly  i had to use   compileSdkVersion 33

Lokesh-Enata, 2024-03-01:
> Unfortunately the flutter [flutter_bluettoth_serial_ble](https://pub.dev/packages/flutter_bluetooth_serial_ble/install) has the same problem. When request location at the first time in android 12 or [...]

No what i have done is if we go and directly run the repo app of this project it will work fine on android 12 but will crash in android 13 so i imported discovery page and connnection page code only from the repo project and i used scoped model and flutter bluetooth serial ble package and that fixed my issue i can connect , send and receive data

Christyansoft, 2024-03-01:
Hi guys, I add all permissions in manifest, updated the permission_handler pacakge and compileSdkVersion is 33.

I just do  Permission.bluetoothConnect.request(); with permission_handler, and automatically, request bluetooth and location. Even if is granted, the app crash at the first time, and shows the error:

E/AndroidRuntime(11660): FATAL EXCEPTION: main
E/AndroidRuntime(11660): java.lang.RuntimeException: Failure delivering result ResultInfo{who=@android:requestPermissions:, request=1451, result=-1, data=Intent { act=android.content.pm.action.REQUEST_PERMISSIONS (has extras) }} to activity {app.MainActivity}: java.lang.NullPointerException: Attempt to invoke interface method 'void io.flutter.plugin.common.MethodChannel$Result.success(java.lang.Object)' on a null object reference
E/AndroidRuntime(11660): 	at android.app.ActivityThread.deliverResults(ActivityThread.java:5375)
E/AndroidRuntime(11660): 	at android.app.ActivityThread.handleSendResult(ActivityThread.java:5414)
E/AndroidRuntime(11660): 	at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:54)
E/AndroidRuntime(11660): 	at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
E/AndroidRuntime(11660): 	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
E/AndroidRuntime(11660): 	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
E/AndroidRuntime(11660): 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2281)
E/AndroidRuntime(11660): 	at android.os.Handler.dispatchMessage(Handler.java:106)
E/AndroidRuntime(11660): 	at android.os.Looper.loopOnce(Looper.java:223)
E/AndroidRuntime(11660): 	at android.os.Looper.loop(Looper.java:317)
E/AndroidRuntime(11660): 	at android.app.ActivityThread.main(ActivityThread.java:7942)
E/AndroidRuntime(11660): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(11660): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:568)
E/AndroidRuntime(11660): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1045)
E/AndroidRuntime(11660): Caused by: java.lang.NullPointerException: Attempt to invoke interface method 'void io.flutter.plugin.common.MethodChannel$Result.success(java.lang.Object)' on a null object reference
E/AndroidRuntime(11660): 	at id.kakzaki.blue_thermal_printer.BlueThermalPrinterPlugin.getBondedDevices(BlueThermalPrinterPlugin.java:462)
E/AndroidRuntime(11660): 	at id.kakzaki.blue_thermal_printer.BlueThermalPrinterPlugin.onRequestPermissionsResult(BlueThermalPrinterPlugin.java:413)
E/AndroidRuntime(11660): 	at io.flutter.embedding.engine.FlutterEngineConnectionRegistry$FlutterEngineActivityPluginBinding.onRequestPermissionsResult(FlutterEngineConnectionRegistry.java:785)
E/AndroidRuntime(11660): 	at io.flutter.embedding.engine.FlutterEngineConnectionRegistry.onRequestPermissionsResult(FlutterEngineConnectionRegistry.java:413)
E/AndroidRuntime(11660): 	at io.flutter.embedding.android.FlutterActivityAndFragmentDelegate.onRequestPermissionsResult(FlutterActivityAndFragmentDelegate.java:801)
E/AndroidRuntime(11660): 	at io.flutter.embedding.android.FlutterActivity.onRequestPermissionsResult(FlutterActivity.java:908)
E/AndroidRuntime(11660): 	at android.app.Activity.dispatchRequestPermissionsResult(Activity.java:8593)
E/AndroidRuntime(11660): 	at android.app.Activity.dispatchActivityResult(Activity.java:8450)
E/AndroidRuntime(11660): 	at android.app.ActivityThread.deliverResults(ActivityThread.java:5368)
E/AndroidRuntime(11660):

zamirszn, 2024-03-01:
> Hi guys, I add all permissions in manifest, updated the permission_handler pacakge and compileSdkVersion is 33.
>
> I just do  Permission.bluetoothConnect.request(); with permission_handler, and [...]

can you share your code , so I can test with my device

Christyansoft, 2024-03-01:
Sorry, i can't, but I just trying requestPermission using:

flutter_bluetooth_serial: ^0.4.0
permission_handler: ^11.3.0
Android 12

Manifest:
`   <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission android:name="android.permission.BLUETOOTH_PRIVILEGED" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.FOREGROUND_SERVICE" />
    <uses-permission android:name="android.permission.WAKE_LOCK" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW" />
    <uses-permission android:name="android.permission.RECEIVE_BOOT_COMPLETED" />
    <uses-permission android:name="android.permission.REQUEST_IGNORE_BATTERY_OPTIMIZATIONS" />
    <uses-permission android:name="android.permission.READ_MEDIA_IMAGES" />
    <uses-permission
        android:name="android.permission.READ_EXTERNAL_STORAGE"
        android:maxSdkVersion="33" />`

like:

`
  final permissionResult = await Permission.bluetoothConnect.request();
`
Just do it at the first time and app will crash

Christyansoft, 2024-03-06:
Just in case, i figure out how fix this, In addition to add bluetooth permissions in manifest, we should  request these bluetooth permissions. like this:

```
     await Permission.bluetoothScan.request();
     await Permission.bluetoothConnect.request();
     await Permission.location.request();
```