edufolly/flutter_bluetooth_serial #212: FlutterBluetoothSerialPlugin.java uses or overrides a deprecated API.

HamzaKlf, 2024-01-29:
Lately, my generated Flutter APK has been experiencing issues with controlling the Android Bluetooth, leading to difficulties in establishing connections.

> Flutter 3.13.0 • channel stable • https://github.com/flutter/flutter.git
> Tools • Dart 3.1.0 • DevTools 2.25.0

while generating the Apk, the following **error** is shown:

> Note: C:\Users\Usuario\AppData\Local\Pub\Cache\hosted\pub.dev\flutter_bluetooth_serial-0.4.0\android\src\main\java\io\github\edufolly\flutterbluetoothserial\FlutterBluetoothSerialPlugin.java uses or overrides a deprecated API.

During the **debugging process**, the following message appears when attempting to disable the Android Bluetooth from within the app:

> I/BluetoothAdapter( 2445): disable()
> I/BluetoothAdapter( 2445): BluetoothAdapter() : com.example.test_reactive
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445): Failed to handle method call
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445): java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for AttributionSource { uid = 10384, packageName = com.example.test_reactive, attributionTag = null, token = android.os.BinderProxy@ff247a7, next = null }: disable
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at android.os.Parcel.createExceptionOrNull(Parcel.java:2437)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at android.os.Parcel.createException(Parcel.java:2421)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at android.os.Parcel.readException(Parcel.java:2404)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at android.os.Parcel.readException(Parcel.java:2346)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at android.bluetooth.IBluetoothManager$Stub$Proxy.disable(IBluetoothManager.java:1046)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at android.bluetooth.BluetoothAdapter.disable(BluetoothAdapter.java:2271)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler.onMethodCall(FlutterBluetoothSerialPlugin.java:593)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at android.os.Handler.handleCallback(Handler.java:938)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at android.os.Handler.dispatchMessage(Handler.java:99)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at android.os.Looper.loopOnce(Looper.java:226)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at android.os.Looper.loop(Looper.java:313)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at android.app.ActivityThread.main(ActivityThread.java:8663)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at java.lang.reflect.Method.invoke(Native Method)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:567)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1135)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445): Caused by: android.os.RemoteException: Remote stack trace:
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at com.android.server.BluetoothManagerService.checkPermissionForDataDelivery(BluetoothManagerService.java:5032)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at com.android.server.BluetoothManagerService.checkConnectPermissionForDataDelivery(BluetoothManagerService.java:5050)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at com.android.server.BluetoothManagerService.checkBluetoothPermissions(BluetoothManagerService.java:1504)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at com.android.server.BluetoothManagerService.disable(BluetoothManagerService.java:1931)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):        at android.bluetooth.IBluetoothManager$Stub.onTransact(IBluetoothManager.java:473)
> E/MethodChannel#flutter_bluetooth_serial/methods( 2445):
> E/flutter ( 2445): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(error, Need android.permission.BLUETOOTH_CONNECT permission for AttributionSource { uid = 10384, packageName = com.example.test_reactive, attributionTag = null, token = android.os.BinderProxy@ff247a7, next = null }: disable, null, java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for AttributionSource { uid = 10384, packageName = com.example.test_reactive, attributionTag = null, token = android.os.BinderProxy@ff247a7, next = null }: disable
> E/flutter ( 2445):      at android.os.Parcel.createExceptionOrNull(Parcel.java:2437)
> E/flutter ( 2445):      at android.os.Parcel.createException(Parcel.java:2421)
> E/flutter ( 2445):      at android.os.Parcel.readException(Parcel.java:2404)
> E/flutter ( 2445):      at android.os.Parcel.readException(Parcel.java:2346)
> E/flutter ( 2445):      at android.bluetooth.IBluetoothManager$Stub$Proxy.disable(IBluetoothManager.java:1046)
> E/flutter ( 2445):      at android.bluetooth.BluetoothAdapter.disable(BluetoothAdapter.java:2271)
> E/flutter ( 2445):      at io.github.edufolly.flutterbluetoothserial.FlutterBluetoothSerialPlugin$FlutterBluetoothSerialMethodCallHandler.onMethodCall(FlutterBluetoothSerialPlugin.java:593)
> E/flutter ( 2445):      at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
> E/flutter ( 2445):      at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
> E/flutter ( 2445):      at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
> E/flutter ( 2445):      at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
> E/flutter ( 2445):      at android.os.Handler.handleCallback(Handler.java:938)
> E/flutter ( 2445):      at android.os.Handler.dispatchMessage(Handler.java:99)
> E/flutter ( 2445):      at android.os.Looper.loopOnce(Looper.java:226)
> E/flutter ( 2445):      at android.os.Looper.loop(Looper.java:313)
> E/flutter ( 2445):      at android.app.ActivityThread.main(ActivityThread.java:8663)
> E/flutter ( 2445):      at java.lang.reflect.Method.invoke(Native Method)
> E/flutter ( 2445):      at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:567)
> E/flutter ( 2445):      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1135)
> E/flutter ( 2445): Caused by: android.os.RemoteException: Remote stack trace:
> E/flutter ( 2445):      at com.android.server.BluetoothManagerService.checkPermissionForDataDelivery(BluetoothManagerService.java:5032)
> E/flutter ( 2445):      at com.android.server.BluetoothManagerService.checkConnectPermissionForDataDelivery(BluetoothManagerService.java:5050)
> E/flutter ( 2445):      at com.android.server.BluetoothManagerService.checkBluetoothPermissions(BluetoothManagerService.java:1504)
> E/flutter ( 2445):      at com.android.server.BluetoothManagerService.disable(BluetoothManagerService.java:1931)
> E/flutter ( 2445):      at android.bluetooth.IBluetoothManager$Stub.onTransact(IBluetoothManager.java:473)
> E/flutter ( 2445):
> E/flutter ( 2445): )
> E/flutter ( 2445): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
[... 61 lines cut ...]
                 to determine the Window background behind the Flutter UI. -->
            <meta-data
[... cut ...]
                <category android:name="android.intent.category.LAUNCHER"/>
            </intent-filter>
        </activity>
        <!-- Don't delete the meta-data below.
             This is used by the Flutter tool to generate GeneratedPluginRegistrant.java -->
        <meta-data
            android:name="flutterEmbedding"
            android:value="2" />
    </application>
</manifest>
```

**build.gradle:**

```
ext.kotlin_version = '1.7.10'
classpath 'com.android.tools.build:gradle:7.3.0'
```

haisamk18, 2024-02-02:
Have you got any solution for this? I am facing this issue while releasing the .apk file

HamzaKlf, 2024-02-02:
So far, there have been no updates for the **flutter_bluetooth_serial** library in the past two years, causing it to become obsolete. Consequently, I've transitioned to using the **flutter_reactive_ble** library, which is functioning well without any issues related to Bluetooth connection or permissions.

haisamk18, 2024-02-03:
But I guess , flutter_reactive_ble doesn’t support Bluetooth Classic .Is
there any alternate of this?

HamzaKlf, 2024-02-05:
Following Dart and Flutter packages can help you add Bluetooth, NFC, Beacon, BLE and other communication feature to your app:
[](https://fluttergems.dev/bluetooth-nfc-beacon/)

Onnesok, 2024-09-22:
I am having same issue.....