edufolly/flutter_bluetooth_serial #220: flutter_bluetooth_serial kotlin not compatible with the latest kotlin

wishwelloklu, 2024-10-14:
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_bluetooth_serial:verifyReleaseResources'.
> A failure occurred while executing com.android.build.gradle.tasks.VerifyLibraryResourcesTask$Action
   > Android resource linking failed
     ERROR:/Users/wishwell/.gradle/caches/transforms-3/ae4b5aebca6216b9e61e6afad4961b83/transformed/core-1.13.1/res/values/values.xml:113:5-122:25: AAPT: error: resource android:attr/lStar not found.

tanmaysuryawanshi, 2024-10-27:
facing the same problem

Sameetdmr, 2024-10-31:
The same problem

wishwelloklu, 2024-11-11:
I haven't found any solution. I have switched to flutter_blue_plus

Shyam-G-Sundhar, 2024-12-10:
Is flutter_blue_plus working fine ?

wishwelloklu, 2024-12-19:
> Is flutter_blue_plus working fine ?

Yes

lastant, 2025-03-26:
For me it helped to go to the source code of the plugin and set `compileSdkVersion 35`