edufolly/flutter_bluetooth_serial #222: flutter_bluetooth_serial:verifyReleaseResources Error

Sameetdmr, 2024-10-31:
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_bluetooth_serial:verifyReleaseResources'.
> A failure occurred while executing com.android.build.gradle.tasks.VerifyLibraryResourcesTask$Action

Zweycinder, 2025-01-14:
hello, did you solve the error ?

Zweycinder, 2025-01-14:
I just found the solution, change is in build.gradle like this 

remove these lines

`
subprojects {
    project.buildDir = "${rootProject.buildDir}/${project.name}"
}
subprojects {
    project.evaluationDependsOn(":app")
}
`

and add these 

subprojects {
    afterEvaluate { project ->
        if (project.extensions.findByName("android") != null) {
            Integer pluginCompileSdk = project.android.compileSdk
            if (pluginCompileSdk != null && pluginCompileSdk < 31) {
                project.logger.error(
                        "Warning: Overriding compileSdk version in Flutter plugin: "
                                + project.name
                                + " from "
                                + pluginCompileSdk
                                + " to 31 (to work around https://issuetracker.google.com/issues/199180389)."
                                + "\nIf there is not a new version of " + project.name + ", consider filing an issue against "
                                + project.name
                                + " to increase their compileSdk to the latest (otherwise try updating to the latest version)."
                )
                project.android {
                    //noinspection GradleDependency
                    compileSdk 31
                }
            }
        }
    }

    project.buildDir = "${rootProject.buildDir}/${project.name}"
    project.evaluationDependsOn(":app")
}

LuvForAirplanes, 2025-01-30:
Thank-you so much @Zweycinder This is working very well! I am looking forward to version 5.0 where @edufolly says this will be fixed.