edufolly/flutter_bluetooth_serial #227: A problem occurred configuring project ':flutter_bluetooth_serial'

B2infosoft-bhuvnesh, 2025-01-24:
Welcome to the issues page of Flutter Bluetooth Serial library.

 This is a template of issue for bugs, prepared to avoid some further questions. 

 mespace not specified. Specify a namespace in the module's build file. See https://d.android.com/r/tools/upgrade-assistant/set-namespace for information about setting the namespace.

     If you've specified the package attribute in the source AndroidManifest.xml, you can use the AGP Upgrade Assistant to migrate to the namespace value in the build file. Refer to https://d.android.com/r/tools/upgrade-assistant/agp-upgrade-assistant for general information about using the AGP Upgrade Assistant.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

edufolly, 2025-01-24:
I'm working on a new version to fix the main issues.

closed 2025-01-24 by edufolly

B2infosoft-bhuvnesh, 2025-01-27:
> I'm working on a new version to fix the main issues.

thanks for replying, i am waiting for your new update.

Rick-OGSS, 2025-03-23:
Thats great, I have the same issue.  Thank you!